kwindrem/SetupHelper #101: SetupHelper on 3.50 GUI_V2

ttraxx, 2024-10-31:
Hello.
Is there a way to run setuphelper and then remote GPIO on v3.50 on GUI_V2?

kwindrem, 2024-10-31:
Remote GPIO is not my package but as I recall, it relies on GuiMods which enhances the gui-v1 but not gui-v2 so you would not see the additional relays and digital inputs in the GUI.

Also, PackageManager adds menus to gui-v1 but not gui-v2 so you would not be able to access those menus to manage the other packages.

I have not seen the v3.50 release so do not know if it allows switching between gui-v1 and gui-v2 like the beta versions did. If so, you can switch to gui-v1 for PackageManager and any other configuration items needed for other packages then switch back to gui-v2.

endurance1968, 2024-11-02:
the release 3.5 is out - it would be good to fully support this as work on v1 is stopped and future is only v2.
I just switch and the only missing I found was the packagemanager menu. Using the full webbrower, far faster navigation I like V2.

ttraxx, 2024-11-02:
> i d

Hello.
3,50 umí přejít na V1, ale guimody na tomto stabilním FW nefungují. Hlásí nekopletní soubory v packake manageru.
I have already modified the automation so that it does not rely on remote GPIO and sends data to tcpmodbus directly, but of course it is not visible in the GUI.

kwindrem, 2024-11-02:
The only way to access PackageManager is to switch the GUI back to v1 which is currently possible. For things like the generator enhancements, you can switch to gui-v1 for configuration then switch back to gui-v2. But obviously the menus will  be the stock gui-v2 menus without any of the GuiMods changes.

The fate of all of my packages is uncertain until there is a path to modify gui-v2.

kwindrem, 2024-11-19:
Most HDMI displays do not provide a mechanism to dim or turn off. This can often be done via the display's menu system but not remotely over HDMI. It's a limitation of the HDMI protocol unfortunately.

I forget the specifics but you can cause the screen blanking to make the screen to all black but the backlight remains on so the screen will glow.