kwindrem/SetupHelper #102: PackageManager keeps restarting

dsrawding, 2024-11-03:
Saw there was an update to guimods so I updated to release v3.50 and PackageManager keeps restarting.  Not able to go back to the old beta firmware so I installed 3.51~2  but still having the same issue.  I uninstalled SetupHelper and reinstalled a number of times and even redownloaded and installed again from source with the same results.  Any ideas?

2024-11-03 02:11:43.179155500 WARNING:>>>> PackageManager v8.22 starting
2024-11-03 02:11:44.751794500 WARNING:starting boot-time reinstall
2024-11-03 02:11:48.208000500 Exception in thread Thread-1:
2024-11-03 02:11:48.208007500 Traceback (most recent call last):
2024-11-03 02:11:48.208010500   File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
2024-11-03 02:11:48.208014500     self.run()
2024-11-03 02:11:48.208016500   File "/data/SetupHelper/PackageManager.py", line 2390, in run
2024-11-03 02:11:48.208020500     self.updateGitHubVersion (packageName, user, branch)
2024-11-03 02:11:48.208023500   File "/data/SetupHelper/PackageManager.py", line 2208, in updateGitHubVersion
2024-11-03 02:11:48.208127500     package.SetGitHubVersion (gitHubVersion)
2024-11-03 02:11:48.208131500   File "/data/SetupHelper/PackageManager.py", line 1397, in SetGitHubVersion
2024-11-03 02:11:48.208135500     self.GitHubVersionNumber = VersionToNumber (version)
2024-11-03 02:11:48.208139500   File "/data/SetupHelper/PackageManager.py", line 455, in VersionToNumber
2024-11-03 02:11:48.208144500     versionNumber += int (numberParts [numberPartsLength])
2024-11-03 02:11:48.208172500 ValueError: invalid literal for int() with base 10: ''
2024-11-03 02:11:49.248301500 WARNING:waiting to aquire lock mainLoop 1
2024-11-03 02:11:49.750363500 WARNING:waiting to aquire lock mainLoop 1
2024-11-03 02:11:50.252074500 WARNING:waiting to aquire lock mainLoop 1
2024-11-03 02:11:50.348458500 WARNING:waiting to aquire lock AddRemove run
2024-11-03 02:11:50.753832500 WARNING:waiting to aquire lock mainLoop 1
2024-11-03 02:11:50.851320500 WARNING:waiting to aquire lock AddRemove run
2024-11-03 02:11:51.260093500 WARNING:waiting to aquire lock mainLoop 1
2024-11-03 02:11:51.353991500 WARNING:waiting to aquire lock AddRemove run
2024-11-03 02:11:51.762497500 WARNING:waiting to aquire lock mainLoop 1
2024-11-03 02:11:51.856472500 WARNING:waiting to aquire lock AddRemove run
2024-11-03 02:11:52.264043500 WARNING:waiting to aquire lock mainLoop 1
2024-11-03 02:11:52.358819500 WARNING:waiting to aquire lock AddRemove run
2024-11-03 02:11:52.765827500 WARNING:waiting to aquire lock mainLoop 1
2024-11-03 02:11:52.861110500 WARNING:waiting to aquire lock AddRemove run
2024-11-03 02:11:53.267520500 WARNING:waiting to aquire lock mainLoop 1
2024-11-03 02:11:53.363293500 WARNING:waiting to aquire lock AddRemove run
2024-11-03 02:11:53.769887500 CRITICAL:timeout waiting for lock mainLoop 1 - restarting PackageManager

dsrawding, 2024-11-03:
Looks like it was an issue with GuiMods.  It has been resolved

closed 2024-11-03 by dsrawding

kwindrem, 2024-11-03:
The issue was a typo in the version file for GuiMods. I fixed that and notices a couple of other issues that got fixed also. One was a white screen which for some reason I didn't see in my testing of v10.68, etc.