kwindrem/SetupHelper #103: Touchscreen calibration on 1024x600 px display (MPI7006)

rholze, 2024-11-18:
Hello, I am having problems setting my 7” display (MPI7006) and the touchscreen together correctly. I have configured the HDMI driver on the console with /data/Rpi/DisplaySetup/setup. Display and touchscreen worked fine together. But of course the correct resolution of 1024x600 pixels was not set. That's why I inserted the lines from the display's display in config.txt (http://www.lcdwiki.com/7inch_HDMI_Display-H)! Then a reboot and the resolution was correct. But now the touchscreen reacted, but you could no longer get into the MENU. Apparently something had been moved here. So I went back to the configuration and “only” called up the touchscreen calibration. But now the resolution was deleted from the config.txt again. How can I set the touchscreen resolution to 1024x600?
Thanks Ronald
[image]
[image]
[image]
[image]

kwindrem, 2024-11-18:
It appears RpiDisplaySetup script is modifying config.txt even when you run the touchscreen calibration only. This is what is undoing your manual changes. I need to fix this but in the mean time, I think you can specify your unit's configuration with a custom display configuration:

at the Select the HDMI mode from the above list (A / P / C): prompt, enter C

then specify 1024 for horizontal resolution, 600 for vertical resolution, 60 for refresh rate and 6 (15:9) for aspect

This should create the hdmi_cvt line per your unit's documentation.

After doing this, you will need to run touch screen calibration which will require a couple of reboots.

kwindrem, 2024-11-19:
I updated the script to allow for running the touchscreen calibration without stepping on config.txt.

Still, I recommend using the custom option to enter your configuration. It is then remembered and will be available in the future if for some reason config.txt gets changed. Firmware updates won't generally touch config.txt but still ...

closed 2024-11-19 by kwindrem