kwindrem/SetupHelper #109: Firmware version checks for all package installs

kwindrem, 2025-01-26:
I am in the process of making package installs more restrictive. In order for the package to install, the author must have tested it on a firmware version and added that specific firmware version to a valid firmware version list. No longer will packages attempt to create file sets during the install process. This is an unfortunate change made necessary by the difficult recovery when the GUI presents a white screen since it is then impossible to access PackageManager.

You will be able to bypass valid version checks but only from the command line.

This also means that a package will not install on new versions until the author adds the new version to the list !!!!

AlbertronicDev, 2025-01-26:
Hi Kevin,
Does this mean that we need to actively update a file containing the "tested on firmware version x"? And if so, wouldn't that mean that if somebody updates to the latest Victron VenusOS without updating the package (or when not having internet access), the package does not get reinstalled?

kwindrem, 2025-01-27:
Yes, what you say is all true. This is mainly to prevent the GUI from crashing (white screen) so maybe if the package does not modify the GUI files, you don't need the valid versions file in the package.

Needs more thought. Thanks for your input.

kwindrem, 2025-01-27:
I an NOT going to implement these changes based on the above comments