kwindrem/SetupHelper #111: Two "Package Manager" menu entries

davinsoftware, 2025-04-09:
I updated SetupHelper from 5.x to 8.31 (latest) and now I have two "Package Manager" menu entries (which appear to lead to identical screens/menus). 

It's not a big deal but a bit weird. I updated the software via the Active Packages install menus. I'm guessing the new SetupHelper installed a second copy. Can I remove the old one? How would I do so?   (given that only one SetupHelper displays in Active Packages). 

[image]

[image]

kwindrem, 2025-04-09:
This is most likely due to the changes to support multiple packages modifying the same file.

The easiest recovery would be to uninstall SetupHelper via the Package Manager menu then do a SetupHelper blind install.

davinsoftware, 2025-04-12:
Thank you, that worked, I'm back to one package manager menu entry.

closed 2025-04-12 by davinsoftware