kwindrem/SetupHelper #114: White Screen loading Gui-V1 in v3.60~75

Albarge, 2025-05-19:
Hi, I had a white screen after testing the Venus beta and switching to Gui-v1, it seemed like packageManager and BatteryAggregator didn't load up as I didn't see them in the device menu then the Quattro shut off power. I managed to reload v3.55 but the white screen is still there for Gui-v1

Ekrano, Venus OS reinstall a few days ago, only running BatteryAggregator and Dbus-SerialBattery mods.

root@ekrano:~# cat /var/log/start-gui/* | tai64nlocal

`2025-05-19 21:22:08.682831500 *** starting start-gui ***
2025-05-19 21:22:08.704818500 method return time=1747689728.704624 sender=:1.32 -> destination=:1.113 serial=3831 reply_serial=2
2025-05-19 21:22:08.704828500    int32 0
2025-05-19 21:22:08.745517500 *** headless device=0
2025-05-19 21:22:08.745818500 *** Waiting for localsettings...
2025-05-19 21:22:08.760014500 *** Localsettings is up, continuing...
2025-05-19 21:22:08.787446500 *** Starting gui, with VNC enabled)
2025-05-19 21:22:08.997637500 QVNCServer created on port 5900
2025-05-19 21:22:09.022656500 Reloading input devices: "LinuxInput:/dev/input/event0 LinuxInput:/dev/input/event1 LinuxInput:/dev/input/event2" 
2025-05-19 21:22:09.028997500 "using /etc/venus for runtime features" 
2025-05-19 21:22:09.029429500 "running on ekrano" 
2025-05-19 21:22:09.041412500 Connecting to deprecated signal QDBusConnectionInterface::serviceOwnerChanged(QString,QString,QString)
2025-05-19 21:22:09.204997500 Creating settings 
2025-05-19 21:22:09.365374500 [VeQItemExportedDbusServices] Registered service "debug.victronenergy.gui" 
2025-05-19 21:22:09.819838500 language is "en" 
2025-05-19 21:22:09.825840500 accessLevelChanged:  2 
2025-05-19 21:22:09.825967500 setAccessLevel:  2 
2025-05-19 21:22:09.856681500 LocalThemeDaemonClient: Looking for assets in "/opt/victronenergy/themes/ccgx" 
2025-05-19 21:22:09.857130500 Screen: 1024x600 150x90mm 173x169dpi
2025-05-19 21:22:10.035929500 TimeZone::valueChanged:  "Europe/London" 
2025-05-19 21:22:10.036144500 ScreenSaver::disable 
2025-05-19 21:22:10.036186500 ScreenSaver::screenOn 
2025-05-19 21:22:10.036612500 ScreenSaver::dbusDisplayOffTimeChanged:  0 
2025-05-19 21:22:10.036726500 ScreenSaver::enable 
2025-05-19 21:22:10.036761500 ScreenSaver::screenOn 
2025-05-19 21:22:12.717440500 file:///opt/victronenergy/gui/qml/main.qml:12:15: Type PageMain unavailable 
2025-05-19 21:22:12.717451500      	initialPage: PageMain {} 
2025-05-19 21:22:12.717454500      	             ^ 
2025-05-19 21:22:12.718433500 file:///opt/victronenergy/gui/qml/PageMain.qml:40:26: Type PageSettings unavailable 
2025-05-19 21:22:12.718442500      				subpage: Component { PageSettings {} } 
2025-05-19 21:22:12.718444500      				                     ^ 
2025-05-19 21:22:12.719088500 file:///opt/victronenergy/gui/qml/PageSettings.qml:193:25: Type PageSettingsPackageManager unavailable 
2025-05-19 21:22:12.719095500      			subpage: Component { PageSettingsPackageManager {} } 
2025-05-19 21:22:12.719098500      			                     ^ 
2025-05-19 21:22:12.719609500 file:///opt/victronenergy/gui/qml/PageSettingsPackageManager.qml:3:1: module "QtQuick" version 2.0 is not installed 
2025-05-19 21:22:12.719615500      import QtQuick 2 
2025-05-19 21:22:12.719617500      ^ 
`

That's not the whole log, but looks like packagemanager unavailable during the initial white screen.

kwindrem, 2025-05-20:
The error in the gui log is that QtQuick 2 is not loaded.

~49 is a pretty old beta and something I can't test because it no longer exists

I can't speak to Mr Manuel's suggestion.

Things do work with v3.55

SetupHelper does modify PageMain via patching, but also replaces QtQuick 1.1 with QtQuick for firmware greater than v3.60~18 which is when Victron made that switch.

It would be interesting to see what is in the PackageManager log because that modification is logged.

I would do a manual uninstall of SetupHelper:

/data/SetupHelper/setup uninstall

Then do a SetupHelper blind install with the most recent version if you aren't already running it.

If you have white screens without SetupHelper installed, then do a firmware update to the latest release or the latest beta. Don't use the backup firmware selection because those may already include corrupted files.

Albarge, 2025-05-20:
Yeah v3.55 fresh installs work, but testing the beta's with SetupHelper so far gives me a white screen in gui-v1

The PackageManager log:

2025-05-19 21:20:31.596601500 *** CCGX booted (0) ***
2025-05-19 21:20:39.695966500 WARNING:>>>> PackageManager v8.33 starting
2025-05-19 21:20:50.175077500 WARNING:starting boot-time reinstall
2025-05-19 21:20:52.162759500 WARNING:boot-time reinstall complete
2025-05-19 21:20:52.247720500 WARNING:checking SetupHelper
2025-05-19 21:20:53.766145500 SetupHelper/setup: --- starting setup script v8.33 action: CHECK
2025-05-19 21:20:55.168217500 WARNING:checking BatteryAggregator
2025-05-19 21:20:57.022237500 BatteryAggregator/setup: --- starting setup script v3.33 action: CHECK
2025-05-19 22:23:07.125387500 SetupHelper/reinstallMods: reinstallMods starting
2025-05-19 22:23:07.165160500 SetupHelper/reinstallMods: reinstallMods finished
2025-05-19 22:23:09.747063500 *** CCGX booted (0) ***
2025-05-19 22:23:18.485312500 WARNING:>>>> PackageManager v8.33 starting
2025-05-19 22:23:29.832145500 WARNING:starting boot-time reinstall
2025-05-19 22:23:31.787486500 WARNING:boot-time reinstall complete
2025-05-19 22:23:31.804636500 WARNING:checking SetupHelper
2025-05-19 22:23:33.559391500 SetupHelper/setup: --- starting setup script v8.33 action: CHECK
2025-05-19 22:23:34.944507500 WARNING:checking BatteryAggregator
2025-05-19 22:23:36.559046500 BatteryAggregator/setup: --- starting setup script v3.33 action: CHECK
2025-05-19 22:38:35.296302500 WARNING:stopping threads
2025-05-19 22:38:35.383155500 WARNING:shutting down com.victronenergy.packageManager dbus service
2025-05-19 22:38:54.089172500 SetupHelper/reinstallMods: reinstallMods starting
2025-05-19 22:38:54.299635500 SetupHelper/reinstallMods: remounting read-write root and resizing - 61 MB now available
2025-05-19 22:38:54.320468500 SetupHelper/reinstallMods: installing PackageManager service - PackageManager will reinstall all packages
2025-05-19 22:38:54.328376500 SetupHelper/reinstallMods: installing PackageManager service
2025-05-19 22:38:54.364611500 SetupHelper/reinstallMods: services manager (svscan) not yet up - PackageManager should start automatically later
2025-05-19 22:38:54.375270500 SetupHelper/reinstallMods: reinstallMods finished
2025-05-19 22:38:55.701051500 *** CCGX booted (0) ***
2025-05-19 22:39:01.398366500 WARNING:>>>> PackageManager  starting
2025-05-19 22:39:09.225670500 WARNING:starting boot-time reinstall
2025-05-19 22:39:09.253542500 WARNING:installing SetupHelper
2025-05-19 22:39:10.613521500 SetupHelper/setup: --- starting setup script v8.33 action: INSTALL
2025-05-19 22:39:11.192430500 WARNING:boot-time reinstall complete
2025-05-19 22:39:11.942726500 SetupHelper/setup: installing files
2025-05-19 22:39:14.235787500 SetupHelper/setup: installing services
2025-05-19 22:39:14.755324500 SetupHelper/setup: restarting PackageManager logger
2025-05-19 22:39:15.030599500 WARNING:SetupHelper install GUI restart pending
2025-05-19 22:39:15.030614500 WARNING:installing BatteryAggregator
2025-05-19 22:39:16.398292500 BatteryAggregator/setup: --- starting setup script v3.33 action: INSTALL
2025-05-19 22:39:17.524089500 BatteryAggregator/setup: ++ Installing Battery Aggregator service
2025-05-19 22:39:35.152676500 BatteryAggregator/setup: ++ Install Python library velib_python (master) from https://github.com/pulquero/velib_python into /data/BatteryAggregator/ext
2025-05-19 22:39:36.617687500 BatteryAggregator/setup: installing BatteryAggregator service
2025-05-19 22:39:42.084458500 BatteryAggregator/setup: service BatteryAggregator running
2025-05-19 22:39:42.180521500 BatteryAggregator/setup: complete - no errors
2025-05-19 22:39:43.268686500 WARNING:stopping threads
2025-05-19 22:39:43.275566500 WARNING:shutting down com.victronenergy.packageManager dbus service
2025-05-19 22:39:43.337479500 WARNING:finishing up in packageManagerEnd.sh
2025-05-19 22:39:43.388111500 WARNING:>>>> PackageManager exiting
2025-05-19 22:39:43.782561500 SetupHelper/packageManagerEnd.sh: starting packageManagerEnd.sh
2025-05-19 22:39:43.811099500 SetupHelper/packageManagerEnd.sh: >>>> restarting GUI and Package Manager
2025-05-19 22:39:43.853005500 SetupHelper/packageManagerEnd.sh: end packageManagerEnd.sh
2025-05-19 22:39:47.033196500 WARNING:>>>> PackageManager v8.33 starting
2025-05-19 22:39:49.160999500 WARNING:checking SetupHelper
2025-05-19 22:39:50.429538500 SetupHelper/setup: --- starting setup script v8.33 action: CHECK
2025-05-19 22:39:51.297171500 WARNING:checking BatteryAggregator
2025-05-19 22:39:52.540431500 BatteryAggregator/setup: --- starting setup script v3.33 action: CHECK

kwindrem, 2025-05-21:
Nothing of interest in the PackageManager log.

I am not seeing any white screens but don't run the packages you do. I did try installing BatteryAggregator and did not get a white screen. I don't have access to the other package.

I'd start by uninstalling the two packages. If you still get a white screen, try reinstalling SetupHelper.

Then reinstall the packages one at a time.

morphias2004, 2025-11-08:
> I am not seeing any issues with v3.66 and there is nothing in the logs you sent indicating a problem.
> 
> But I see you are using a VERY old version of SetupHelper. You are on v5.18 and it's currently v9.1
> 
> What I suggest is that you uninstall all packages then reinstall the firmware. Switch to the backup firmware then do an on-line update to the latest release or beta version as appropriate for your system. After that, do a SetupHelper blind install using the latest version. Following that, you can use PackageManager to update the other packages to their latest version.

I ran a command line uninstall of SetupHelper, deleted the folder and manually downloaded and installed 9.1 and had the same issue.

The same day, I did this on a Cerbo GX and the Ekrano. The Cerbo GX was fine, but the Ekrano is not. Both are on SetupHelper 9.1 and GUIMods 10.97.

kwindrem, 2026-05-27:
Noted. This is an issue with this package specifically and not a generic SetupHelper problem. You should contact the author of RemoteGPIO and have them resolve the issue. I am not sure of the current author. The latest I was involved with is authored by Lucifer06 but the latest update was 11 months ago and I don't see any of those files in that repo.

Sorry I can not be of more help

closed 2026-06-27 by kwindrem