kwindrem/SetupHelper #116: RpiDisplaysetup

pa5cw, 2025-05-25:
I try to setup rpi display, but I get Aanerror.

Setuphelper is installed (version 8.33), 
I see rpidisplaysetup v 3.16,, it says: no package.

When I say download and proceed, it says: could not access rpidisplaysetup.

I used usb stick to install venusdata.tgz (in RPI) and then reboot.

Do I miss something out?

kEES

kwindrem, 2025-05-25:
For each package Package Manager shows three versions: one for what is found on GitHub, one that is “stored” on the system and the version that is Installed. Let me know which you see

You need the system connected to the internet ti see GitHub versions and to download.

venusdata.tgz only contains SetupHelper

Also let me know how you are trying to install RpiDisplaySetup

pa5cw, 2025-05-25:
I only downliaded venusdata.tgz, so that is the problem
I thought that it contains also rpidisplaysetup.

kwindrem, 2025-05-25:
If your GX device doesn’t have internet access, you can download each package from GitHub and place each file on a USB stick. Package Manager will transfer from there. 

Should be instructions in the Setup Helper ReadMe