kwindrem/SetupHelper #117: update Venus os to v3.60 results in white screen

drtinaz, 2025-06-19:
as the title states, updated from 3.55 to 3.60 and get a white screen on gui v1. No guimods installed, just have setup helper v8.33 installed with a few of my own services installed using setup helper.

drtinaz, 2025-06-20:
My packages do not make any changes to the gui. None the less, I have tried reverting to 3.55, uninstalled everything except packagemanager, performed a fresh install to 3.60, and still have a white screen on gui v1. I checked the logs and there are some syntax errors in the package manager script.

I don't have access to my computer at the moment, only my phone and I can't figure out how to attach a file through the GitHub app. Here is the snippet of the log. Hope this helps.

`4000000068554b0517e8b07c WARNING:>>>> PackageManager exiting
@4000000068554b0603dc4224 SetupHelper/packageManagerEnd.sh: starting packageManagerEnd.sh
@4000000068554b0605c28684 SetupHelper/packageManagerEnd.sh: >>>> restarting GUI and Package Manager
@4000000068554b060c15c5dc SetupHelper/packageManagerEnd.sh: end packageManagerEnd.sh
@4000000068554b070eb1f874 /data/SetupHelper/PackageManager.py:420: SyntaxWarning: invalid escape sequence '\D'
@4000000068554b070ee053a4   numberParts = re.split ('\D+', version)
@4000000068554b0710670efc /data/SetupHelper/PackageManager.py:421: SyntaxWarning: invalid escape sequence '\d'
@4000000068554b0710673224   otherParts = re.split ('\d+', version)
`

kwindrem, 2025-06-20:
I looked into the error and it's a syntax warning but everything is working. firmware v3.60 upgrades python over what's in v3.55 which is way you are not seeing the warning with v3.55.

The fact that you are not seeing a stored version for SetupHelper suggests /data/SetupHelper might be corrupt (missing or garbled version file at least).

I am not seeing missing version numbers with v3.60 on my systems.

If you have not done so already, you might try downloading SetupHelper again from GitHub. You should be able to do this from the PackageManager menu while running 3.55

drtinaz, 2025-06-20:
Just to clarify, the stored version number that I made reference to not seeing is the venus os firmware, and not setup helper. I was referring to not being able to switch back to venus 3.60 after reverting back to venus 3.55. in order to get back to venus 3.60 from 3.55 I have to do a new install and am not able to switch back to 3.60 by just doing a backup firmware swap. 

I will revert back to 3.55 again and redownload setup helper and reinstall, then upgrade to 3.60 venus again and see if there is any change.

drtinaz, 2025-06-20:
I did another fresh install of 3.60 and then reinstalled setup helper and had the same problem. I then uninstalled and deleted the setup helper folder in its entirety. Redownloaded and installed from terminal and gui v1 is back. Something must have been corrupted and for whatever reason just redownloading without deleting the folder first didn't fix it. Strange indeed but it is working now.

closed 2025-06-20 by drtinaz

kwindrem, 2025-06-20:
Extremely odd and sorry you experienced the problems.

I know with auto updates, if the GitHub and stored versions match the file isn't downloaded, but I think if you do a manual Download from the PackageManager menu it should do the download regardless of versions.

So ?????