kwindrem/SetupHelper #118: No Uninstall option

thespinmaster, 2025-07-03:
Hi, I'm in the process of researching using the awesome Setuphelper for a service I've written and noticed a *possible* issue.

I installed the Tailscale package to see where files ended up. All good so far.
Then when I went to uninstall, there was no option to uninstall. I just see the "Previous" "Next" and "Download" buttons.

If I click on the "Download" button, it redownloads, and the "Install" and "Uninstall" buttons then appear.
If I move back out of the Tailscale page and back into the "Active Packages">>Tailscale page, the "Install" and "Uninstall" buttons have disappeared again.

This appears to be true of all packages not just Tailscale, so assuming the problem is with the Setuphelper.

System Info:
Cerbo GX MK2
Venus OS (Latest BETA 3.70~5)

Issue is not huge as the workaround is to click the download button, then uninstall, + the Venus beta may be the issue.

Additional info, that may help
I installed Setuphelper then GuiMods
"Auto Install Packages" is checked
Then upgraded the Venus OS from the stable release (3.6...) of Venus OS to the 3.7 BETA
Then installed Tailscale, and then noticed the issue.

[image]

kwindrem, 2025-07-04:
I see the problem. It turns out the Install and Uninstall buttons are there, but on a separate line underneath Download. If you drag up on the screen the buttons will appear and line up with Download. Unfortunately there's no way to do this with the remote GUI and the arrow buttons but a click-drag with the mouse does work.

Not sure what's going on but I'll look into it.

thespinmaster, 2025-07-04:
Arh yes, I see them. Thanks :)

kwindrem, 2025-07-13:
This should now be fixed. Not sure why it was happening.

The side effect of the change is that the status message is part of the scrolling to clickable items with the "hard buttons" (CCGX and web-based GUI (v1)).

closed 2025-07-13 by kwindrem