kwindrem/SetupHelper #122: Stop RUUVI sensors battery level display?

XeroX6969, 2025-07-20:
Hi Kevin,
I have been unable to find how to prevent the Ruuvi temp sensor battery level display on the main screen ? This is quite recent I find it make temp reading difficult.
Sorry to be basic as compare to other issues around the site.
Thanks for this great work.
Francois

kwindrem, 2025-07-20:
Where are you seeing battery level? What GUI (classic or new)? Are you running GuiMods?
Which sensor model? Firmware and package versions?

XeroX6969, 2025-07-21:
I use the classic GUI (v3.63), I'm running GuiMods (10.95 setup 8.34) and 4 Ruuvi (3 internals and one spécific outside). Up to recently , only the 4 temps were shown, but now the battery levels for 3 of them (internals I suppose)is shown: 

[image]

[image]

Thanks for your help
Francois