kwindrem/SetupHelper #124: Installation of SetupHelper fails

hussl1, 2025-07-31:
Hi there, 

I am trying to get started with SetupHelper but don't get very far...

`
root@einstein:~# wget -qO - https://github.com/kwindrem/SetupHelper/archive/latest.tar.gz | tar -xzf - -C /data
root@einstein:~# mv -f /data/SetupHelper-latest /data/SetupHelper
root@einstein:~# /data/SetupHelper/setup
--- starting setup script v7.9 action NONE
ERROR: patch unsuccesful for PageSettings.qml
restarting generator service
svc: warning: unable to chdir to /service/dbus-generator-starter: file does not exist
complete - no changes were made
`

Firmware is 3.60 on a CerboGX.
Any ideas?
Thanks for your help.

hussl1, 2025-08-01:
That worked, thanks!

kwindrem, 2025-08-01:
Great. I'll update the ReadMe. 

Most people use the blind install procedure so the wget doesn't get much use.