kwindrem/SetupHelper #128: Update

marc115uk, 2025-08-25:
Possibly a stupid question but how to update setup helper, all the other packages show what is latest version on GitHub and gives an option to update whereas no version showing for setup helper, currently on 8.22 have noticed now upto v9. Running venous v3.55 

Many thanks

marc115uk, 2025-08-25:
[image]

kwindrem, 2025-08-25:
Odd. The GitHub version should show up when the others do.

Go into SetupHelper from the Active packages menu you show above and make sure the GitHub user and version are correct:

kwindrem
latest

If that doesn't work, try initializing PackageManager from Settings / PackageManager / Restart or initialize ...

BTW, GitHub version information "expires" after 10 minutes so stale information is not displayed. If you remain in the Active packages menu or the Package edit menu for this long and the GitHub check frequency is longer than this, you will see GitHub version information go away. Reentering those menus or switching to a different package (Package edit menu) will refresh version info.

(Some users with limited internet access wanted to minimize the frequency of requesting GitHub info from the web. Which is why the GitHub frequency check selection exists.)

marc115uk, 2025-08-25:
Thanks, it was the latest tag, was set to 6.2 fairly sure I wouldn't have changed it after installing it a few years back, had changed it to 9.0 today when I noticed version 9 was the latest, before posting this, but didn't work. Changed to latest now and it popped up with 9 straight away 

However new issue now I'm on version 9 I am unable to get into any of the pages in active packages to update change enable disable etc. Have tried to get into the pages on inactive packages and that works, but when I go back to active packages unable to open them. Tried rebooting Venus, tried initialising package manager and restarting package manager but still unable to open them

closed 2025-08-25 by kwindrem